StanleyInnovation/robotiq_85_gripper #1: Gripper does not spawn correctly in Gazebo

J-Schaefer, 2015-11-30:
When I spawn the gripper in Gazebo with an UR5, the joints rotate uncontrolled -- they follow gravity.
The only joint which doesn't, are the fixed ones and the left_knuckle_joint.
The gripper is also not controllable in Gazebo with the joint_state_publisher.

[image]

I loaded the gripper inside the ur5_robot.urdf.xacro-file from Universal Robots.
When I start Rviz the gripper is highlighted in red.

How can I correctly connect the gripper to a robotic arm? Is my way to connect the gripper to the UR5 not correct?

PatrickHussey, 2015-11-30:
Hi the gazebo model I did for the gripper is not included in the driver. Gazebo requires the under -actuated joints to be modeled differently for simulation as it does not handle them gracefully. You need a physical gripper to use this driver for now. I plan on eventually wrapping the gripper sim model in but probably not immediately.

J-Schaefer, 2015-12-02:
I solved the problem by adding the joints to the ur5.transmission.xacro and the controller:

**ur5.transmission.xacro:**

    <?xml version="1.0"?>
    <robot xmlns:xacro="http://www.ros.org/wiki/xacro">
      <xacro:macro name="ur5_arm_transmission" params="prefix">

        [I skipped the code that has been here before]

        <transmission name="${prefix}robotiq_85_left_knuckle_trans">
          <type>transmission_interface/SimpleTransmission</type>
          <joint name="${prefix}robotiq_85_left_knuckle_joint">
            <hardwareInterface>PositionJointInterface</hardwareInterface>
         </joint>
          <actuator name="${prefix}robotiq_85_left_knuckle_motor">
            <mechanicalReduction>1</mechanicalReduction>
          </actuator>
        </transmission>

        <transmission name="${prefix}robotiq_85_right_knuckle_trans">
          <type>transmission_interface/SimpleTransmission</type>
          <joint name="${prefix}robotiq_85_right_knuckle_joint">
            <hardwareInterface>PositionJointInterface</hardwareInterface>
          </joint>
          <actuator name="${prefix}robotiq_85_right_knuckle_motor">
            <mechanicalReduction>1</mechanicalReduction>
          </actuator>
        </transmission>

        <transmission name="${prefix}robotiq_85_left_inner_knuckle_trans">
          <type>transmission_interface/SimpleTransmission</type>
          <joint name="${prefix}robotiq_85_left_inner_knuckle_joint">
            <hardwareInterface>PositionJointInterface</hardwareInterface>
          </joint>
          <actuator name="${prefix}robotiq_85_left_inner_knuckle_motor">
            <mechanicalReduction>1</mechanicalReduction>
          </actuator>
        </transmission>

        <transmission name="${prefix}robotiq_85_right_inner_knuckle_trans">
          <type>transmission_interface/SimpleTransmission</type>
          <joint name="${prefix}robotiq_85_right_inner_knuckle_joint">
            <hardwareInterface>PositionJointInterface</hardwareInterface>
          </joint>
          <actuator name="${prefix}robotiq_85_right_inner_knuckle_motor">
            <mechanicalReduction>1</mechanicalReduction>
          </actuator>
        </transmission>

        <transmission name="${prefix}robotiq_85_left_finger_tip_trans">
          <type>transmission_interface/SimpleTransmission</type>
          <joint name="${prefix}robotiq_85_left_finger_tip_joint">
            <hardwareInterface>PositionJointInterface</hardwareInterface>
          </joint>
          <actuator name="${prefix}robotiq_85_left_finger_tip_motor">
            <mechanicalReduction>1</mechanicalReduction>
          </actuator>
        </transmission>

        <transmission name="${prefix}robotiq_85_right_finger_tip_trans">
          <type>transmission_interface/SimpleTransmission</type>
          <joint name="${prefix}robotiq_85_right_finger_tip_joint">
            <hardwareInterface>PositionJointInterface</hardwareInterface>
          </joint>
          <actuator name="${prefix}robotiq_85_right_finger_tip_motor">
            <mechanicalReduction>1</mechanicalReduction>
          </actuator>
        </transmission>

      </xacro:macro>

</robot>

**arm_controller_ur5.yaml:**

    arm_controller:
      type: position_controllers/JointTrajectoryController
      joints:
         - shoulder_pan_joint
         - shoulder_lift_joint
         - elbow_joint
         - wrist_1_joint
         - wrist_2_joint
         - wrist_3_joint
         - robotiq_85_left_knuckle_joint
         - robotiq_85_right_knuckle_joint
         - robotiq_85_left_inner_knuckle_joint
         - robotiq_85_right_inner_knuckle_joint
         - robotiq_85_left_finger_tip_joint
         - robotiq_85_right_finger_tip_joint
      constraints:
          goal_time: 0.6
          stopped_velocity_tolerance: 0.05
          shoulder_pan_joint: {trajectory: 0.1, goal: 0.1}
          shoulder_lift_joint: {trajectory: 0.1, goal: 0.1}
          elbow_joint: {trajectory: 0.1, goal: 0.1}
          wrist_1_joint: {trajectory: 0.1, goal: 0.1}
          wrist_2_joint: {trajectory: 0.1, goal: 0.1}
          wrist_3_joint: {trajectory: 0.1, goal: 0.1}
          robotiq_85_left_knuckle_joint: {trajectory: 0.1, goal: 0.1}
          robotiq_85_right_knuckle_joint: {trajectory: 0.1, goal: 0.1}
          robotiq_85_left_inner_knuckle_joint: {trajectory: 0.1, goal: 0.1}
          robotiq_85_right_inner_knuckle_joint: {trajectory: 0.1, goal: 0.1}
          robotiq_85_left_finger_tip_joint: {trajectory: 0.1, goal: 0.1}
          robotiq_85_right_finger_tip_joint: {trajectory: 0.1, goal: 0.1}
      stop_trajectory_duration: 0.5
      state_publish_rate:  25
      action_monitor_rate: 10

This works quite fine. Thank you for your appreciation.

MetaMemoryT, 2016-02-10:
I am trying to replicate your solution @J-Schaefer,

https://github.com/ros-industrial/universal_robot/blob/hydro-devel/ur_description/urdf/ur5.transmission.xacro only exists in hydro-devel branch

in indigo it was replaced with https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_description/urdf/ur.transmission.xacro

any hints to make this work in indigo?

how did you add the gripper to https://github.com/ros-industrial/universal_robot/blob/indigo-devel/ur_description/urdf/ur5_robot.urdf.xacro ?

J-Schaefer, 2016-02-15:
Hello,
I'm using Indigo but with an older version of the UR-indigo-branch. 

I attached the gripper to the UR5 by writing this in the file _ur5_robot.urdf.xacro_:

```
<xacro:include filename="$(find robotiq_85_description)/urdf/robotiq_85_gripper.xacro" />
[...]
<joint name="Gripper_Adapter_Fix" type="fixed">
 <parent link="fts_toolside" />
 <child link="robotiq_85_root" />
 <origin xyz="0.0 0.0 -.0123" rpy="0.0 ${- pi / 2} 0.0" />
</joint>
```
The link 'fts_toolside' is from the FT150.

I updated the files I already described here:
I deleted all the Joints that are controlled indirectly as mimic-joints. Only the Joint 'robotiq_85_left_knuckle_joint' remains:

_ur5.transmission.xacro_:
```
<?xml version="1.0"?>
<robot xmlns:xacro="http://www.ros.org/wiki/xacro">

  <xacro:macro name="ur5_arm_transmission" params="prefix">

    <transmission name="${prefix}shoulder_pan_trans">
      <type>transmission_interface/SimpleTransmission</type>
      <joint name="${prefix}shoulder_pan_joint">
        <hardwareInterface>PositionJointInterface</hardwareInterface>
      </joint>
      <actuator name="${prefix}shoulder_pan_motor">
        <mechanicalReduction>1</mechanicalReduction>
      </actuator>
    </transmission>
  
    <transmission name="${prefix}shoulder_lift_trans">
      <type>transmission_interface/SimpleTransmission</type>
      <joint name="${prefix}shoulder_lift_joint">
        <hardwareInterface>PositionJointInterface</hardwareInterface>
      </joint>
      <actuator name="${prefix}shoulder_lift_motor">
        <mechanicalReduction>1</mechanicalReduction>
      </actuator>
    </transmission>
  
    <transmission name="${prefix}elbow_trans">
      <type>transmission_interface/SimpleTransmission</type>
      <joint name="${prefix}elbow_joint">
        <hardwareInterface>PositionJointInterface</hardwareInterface>
      </joint>
      <actuator name="${prefix}elbow_motor">
        <mechanicalReduction>1</mechanicalReduction>
      </actuator>
    </transmission>
  
    <transmission name="${prefix}wrist_1_trans">
      <type>transmission_interface/SimpleTransmission</type>
      <joint name="${prefix}wrist_1_joint">
        <hardwareInterface>PositionJointInterface</hardwareInterface>
      </joint>
      <actuator name="${prefix}wrist_1_motor">
        <mechanicalReduction>1</mechanicalReduction>
      </actuator>
    </transmission>
  
    <transmission name="${prefix}wrist_2_trans">
      <type>transmission_interface/SimpleTransmission</type>
      <joint name="${prefix}wrist_2_joint">
        <hardwareInterface>PositionJointInterface</hardwareInterface>
[... 98 lines cut ...]
		<offset>0.0</offset>
	</plugin>

<!--
    <plugin name="gazebo_ros_power_monitor_controller" filename="libgazebo_ros_power_monitor.so">
      <alwaysOn>true</alwaysOn>
      <updateRate>1.0</updateRate>
      <timeout>5</timeout>
      <powerStateTopic>power_state</powerStateTopic>
      <powerStateRate>10.0</powerStateRate>
      <fullChargeCapacity>87.78</fullChargeCapacity>     
      <dischargeRate>-474</dischargeRate>
      <chargeRate>525</chargeRate>
      <dischargeVoltage>15.52</dischargeVoltage>
      <chargeVoltage>16.41</chargeVoltage>
    </plugin>
-->

  </gazebo>

</robot>
```

Seems like the guys from UR changed a lot.
I hope this will help you.

MetaMemoryT, 2016-02-17:
Thanks @J-Schaefer for the information.

I am trying to set up robotiq_85_gripper in a separate simulation now to simplify things.   I have created a package robotiq_85_gazebo in my branch gazebo see: https://github.com/StanleyInnovation/robotiq_85_gripper/compare/master...MetaMemoryT:gazebo

I currently get the error:
```
[ERROR] [1455667016.392414608, 0.635000000]: Could not find joint 'robotiq_85_left_knuckle_joint' in 'hardware_interface::PositionJointInterface'.
[ERROR] [1455667016.392529475, 0.635000000]: Failed to initialize the controller
[ERROR] [1455667016.392566050, 0.635000000]: Initializing controller 'robotiq_85_controller' failed
Error when loading robotiq_85_controller
[ERROR] [1455667016.458055696, 0.699000000]: Could not start controller with name robotiq_85_controller because no controller with this name exists
```

Thanks again for the help.

PatrickHussey, 2016-02-17:
You need to use mimic joints for everything except the left knuckle and the gazebo mimic plugin. Add the left knuckle to the ROS controller spawned/manager. And set all other joints as passive in the SDF for MoveIt. 

I have a working sim just haven't had time to put it up.

MetaMemoryT, 2016-02-19:
thanks I am using setup assistant now.

MetaMemoryT, 2016-02-20:
I would love to beta test your working sim, even if it wasn't complete.

PatrickHussey, 2016-02-22:
I had a little time so I added the gazebo sim to the package. You just need to run:

roslaunch robotiq_85_gazebo robotiq_85_moveit_sim.launch

It just runs through a bunch of random trajectories to validate the simulation.

Probably need to go back an put in some env variables for the xacro configuration, but since you're just using it for sim shouldn't matter

PatrickHussey, 2016-02-22:
also added the moveit configuration, so should be good to go now. closing

closed 2016-02-22 by PatrickHussey