StanleyInnovation/robotiq_85_gripper #2: Gripper red highlighted in Rviz

J-Schaefer, 2015-12-08:
Hello,

how can I integrate the gripper with an UR5 robot, correctly so that the gripper can be moved in Rviz using MoveIt.
I mounted the gripper to the robot by writing it in the ur5_robot.urdf.xacro.

    <xacro:include filename="$(find robotiq_85_description)/urdf/robotiq_85_gripper.xacro" />

    <joint name="Gripper_Adapter_Fix" type="fixed">
     <parent link="fts_toolside" />
     <child link="robotiq_85_root" />
     <origin xyz="0.0 0.0 +.0123" rpy="0.0 ${- pi / 2} 0.0" />
    </joint>

The gripper spawns correctly in Gazebo, but is highlighted red in Rviz and cannot be moved.

Is there a better way to integrate the gripper?

Best regards
J-Schaefer

MetaMemoryT, 2016-02-11:
:+1: I am stuck on this too

the status inside rviz says:

```
Initialized.
Changed start state
Changed goal state
Goal state colliding links:
ee_link - robotiq_85_base_link
robotiq_85_base_link - robotiq_85_left_inner_knuckle_link
robotiq_85_base_link - robotiq_85_left_knuckle_link
robotiq_85_base_link - robotiq_85_right_inner_knuckle_link
robotiq_85_base_link - robotiq_85_right_knuckle_link
robotiq_85_base_link - wrist_2_link
robotiq_85_base_link - wrist_3_link
robotiq_85_left_finger_link - robotiq_85_left_finger_tip_link
robotiq_85_left_finger_link - robotiq_85_left_knuckle_link
robotiq_85_left_finger_tip_link - robotiq_85_left_inner_knuckle_link
robotiq_85_right_finger_link - robotiq_85_right_finger_tip_link
robotiq_85_right_finger_link - robotiq_85_right_knuckle_link
robotiq_85_right_finger_tip_link - robotiq_85_right_inner_knuckle_link
.
...
more of the above message repeated...

```

PatrickHussey, 2016-02-11:
You need to add pairs that don't need to be collision checked to the SDF file in the moveit configuration or use the setup assistant to generate that automatically.

PatrickHussey, 2016-02-22:
adding gazebo and moveit configurations closing

closed 2016-02-22 by PatrickHussey