StanleyInnovation/robotiq_85_gripper #3: Failed to contact gripper 0....ABORTING

james2254, 2016-01-28:
When i run this cmd " rosrun robotiq_85_driver robotiq_85_driver ",there is a error coming up.
 

[image]

PatrickHussey, 2016-01-28:
Is your username a member of dial out? Are you sure the comport and baud settings are correct? And can you talk to the Gripper with Robotiqs tool, using the default settings?

Just a few things to check. Also try the launch file.

james2254, 2016-01-28:
I can't setting baud in 115200，there are three types baud——125k，250k，500k。I select 125k baud，but it fail too。I can talk to the RUI tool。

PatrickHussey, 2016-01-28:
The default for the robotiq 85 2 finger gripper (the new one not the C model) is 115200, so not sure why it can't be set. This driver only supports the new ones not the old one (C 200). 

You need to set the baud param in the driver to what the gripper is set to. ie

rosrun robotiq_85_driver robotiq_85_driver baud:=125000 (or whatever it actually is). You also need to make sure the port is set correctly it defaults to /dev/ttyUSB0.

For example baud=125k and port=/dev/ttyUSB1

rosrun robotiq_85_driver robotiq_85_driver baud:=125000 port=/dev/ttyUSB1

closed 2016-01-28 by PatrickHussey