StanleyInnovation/robotiq_85_gripper #4: [ WARN] [1455207433.835350875, 923.836000000]: Assimp reports no scene in package://robotiq_85_description/meshes/robotiq_85_gripper/collision/robotiq_85_inner_knuckle_link.stl

MetaMemoryT, 2016-02-11:
I think this has to do with malformed file: http://notebook.updroid.com/2016/01/04/from-solidworks-to-collada-urdf/

PatrickHussey, 2016-02-11:
It's actually an issue with the length of the file path name. The 'solid' word in the header was fixed when I created them.

I'll fix it when I get a chance

MetaMemoryT, 2016-02-11:
Thanks for the info:

I got around the problem by renaming:

robotiq_85_inner_knuckle_link.stl to robotiq_85_ikl.stl

PatrickHussey, 2016-02-11:
Sounds good, I made sure the issue got resolved for everything by renaming 

robotiq_85_description/meshes/robotiq_85_gripper/collision/

to

robotiq_description/meshes/robotiq_85/collision/

PatrickHussey, 2016-02-22:
fixed paths in latest checkin to get rid of this error

closed 2016-02-22 by PatrickHussey