StanleyInnovation/robotiq_85_gripper #8: Failed to load plugin libroboticsgroup_gazebo_mimic_joint_plugin.so

keerthanamanivannan, 2017-02-27:
System Specs:
Ubuntu 14.04.5 trusty 
ROS: Indigo
Gazebo: 2.2.3

I just did the following: Git cloned the repo and built it (there was a cmake error which I fixed) but whenever I tried to launch the robotiq_85 in Gazebo with` roslaunch robotiq_85_gazebo robotiq_85.launch` , it shows this error(x5): 

> Error [Plugin.hh:156] Failed to load plugin libroboticsgroup_gazebo_mimic_joint_plugin.so: libroboticsgroup_gazebo_mimic_joint_plugin.so: cannot open shared object file: No such file or directory

I've spent a lot of time on this, please let me know what I can do to fix it. I've tried it on multiple machines as well. Still the same problem. Please help me with this. Thank you.

PatrickHussey, 2017-02-27:
Did you source the workspace?

source ./devel/setup.bash

At the top of the workspace before running sim.

That plugin gets built in the workspace so ROS won't find it unless you source the workspace first.

keerthanamanivannan, 2017-02-27:
yes I did.

keerthanamanivannan, 2017-02-28:
Ok I seem to have solved the problem. Odd that I had to face such an issue. 
I looked at the other **/build** directories in my other workspaces. They didn't have the built **.so** files in them. After building the catkin package, the built **.so** files end up in the **/devel/lib** directory. But these two didn't. I don't know why. I copied these two files into the **/devel/lib** folder and everything seems to work fine. The gripper loads properly. 

I don't understand why this issue cropped up for me. And I tried it in two other machines as well. Same problem.

keerthanamanivannan, 2017-03-15:
I used **catkin build** from catkin_tools to build the packages. When I tried building the packages with catkin_make, it works perfectly. Possibly there's a problem with catkin_tools and it's failing to link the plugin.so files into the **/devel/lib** folder.

closed 2017-03-15 by keerthanamanivannan

wafaezouhair, 2018-02-28:
hello,
please i want to know what is the command to open/close the robotiq_85 _gripper in gazebo using python