StanleyInnovation/robotiq_85_gripper #16: gripper cannot grasp object in gazebo

yt100323, 2018-01-09:
I attached the gripper model to ur5 and simulated them in Gazebo, the grasped target would slide out of the gripper.

yt100323, 2018-01-09:
my Gazebo is 2.2.3, and in Ubuntu 14.04.5 and ROS indigo

cyrilli, 2018-01-15:
I solved the problem by using Jennifer's [gazebo grasp plugin](https://github.com/JenniferBuehler/gazebo-pkgs).

LuserBen, 2018-03-29:
Hello, cyrilli.
Can you tell me how to edit the file "gzplugin_grasp_fix.urdf.xacro" ? 
Thanks

geweihgg, 2018-05-03:
Hi,
    I’m doing some work with ur5 + robotiq 85, but I don’t know how to open/close the gripper in gazebo.
    Could you please tell me how to control the gripper in gazebo? 
    Thank you very much.
                                                                                                                                GeWei

LuserBen, 2018-05-04:
Hi, @geweihgg 
I use moveit to open/close the gripper. I set the gripper as a planning group and set the joint value by moveit C++ interface.

geweihgg, 2018-05-04:
Hi, @LuserBen 
Thank you very much for your help.
Now I'm using the MoveIt C++ Interface too. 
But I encountered another problem. It seems that the robotiq gripper model collides with the object which I want to grasp. Have you solved this?

whwqy, 2018-09-26:
hi @geweihgg  I also encountered the same problem .Do you have a solution now? thank you

geweihgg, 2018-09-26:
@whwqy , yes, this plugin is not stable as you wish, we can only relieve this issue~

zhekaiTong, 2018-11-14:
Hi, @geweihgg 
I tried to combine ur10 model with robotiq 85, but had two problems:
1. When I plan the gripper motion in moveit, the gripper in gazebo does not move correspondingly. Instead, there is a warning said 

> [WARN] [1542160314.677185, 28.856000]: Controller Spawner couldn't find the expected controller_manager ROS interface.

The warning may be because missing gazebo_ros_control plugin, however, I added the plugin in my urdf:
`<gazebo>
        <plugin name="gazebo_ros_control" filename="libgazebo_ros_control.so">
            <robotNamespace>/ur10_robotiq_85</robotNamespace>
        </plugin>
    </gazebo>`
the warning still comes out.

2. I set the gripper joints as a planning group in moveit via moveit setup assistant. However, moveit said my group is not a chain. Below is my group in .srdf:
`<group name="gripper">
        <joint name="gripper_finger1_inner_knuckle_joint" />
        <joint name="gripper_finger1_finger_tip_joint" />
        <joint name="gripper_finger1_joint" />
        <joint name="gripper_finger1_finger_joint" />
        <joint name="gripper_finger2_inner_knuckle_joint" />
        <joint name="gripper_finger2_finger_tip_joint" />
        <joint name="gripper_finger2_joint" />
        <joint name="gripper_finger2_finger_joint" />
        <joint name="wrist_to_coupler" />
        <joint name="gripper_base_joint" />
    </group>`
I just added a coupler link between robot wrist and the gripper.

Do you have solutions to these problems? Thanks a lot.

LuserBen, 2018-11-14:
Hi, @zhekaiTong  
1. Maybe you should install gazebo_ros_control
$ sudo apt-get install ros-indigo-gazebo-ros-control

2. Maybe you set chain (just set joints is OK) in Planning Groups by moveit setup assistant.

Just I guess.

zhekaiTong, 2018-11-14:
Hi, @LuserBen 
Thanks for your reply.
Actually, I have installed gazebo_ros_control via (I am using ros kinetic+ubuntu 16.04):
$ sudo apt-get install ros-kinetic-gazebo-ros-control

However, the warnings are still there:
> [ INFO] [1542163123.786343749, 0.159000000]: Loading gazebo_ros_control plugin
[ INFO] [1542163123.786437538, 0.159000000]: Starting gazebo_ros_control plugin in namespace: /ur10_robotiq_85
[ INFO] [1542163123.787208705, 0.159000000]: gazebo_ros_control plugin is waiting for model URDF in parameter [/robot_description] on the ROS param server.
[ WARN] [1542163123.899959944, 0.159000000]: Deprecated syntax, please prepend 'hardware_interface/' to 'PositionJointInterface' within the <hardwareInterface> tag in joint 'gripper_finger1_joint'.
[ INFO] [1542163123.905628038, 0.159000000]: Loaded gazebo_ros_control.
[ WARN] [1542163123.905994304, 0.160000000]: The default_robot_hw_sim plugin is using the Joint::SetPosition method without preserving the link velocity.
[ WARN] [1542163123.906015237, 0.160000000]: As a result, gravity will not be simulated correctly for your model.
[ WARN] [1542163123.906049060, 0.160000000]: Please set gazebo_pid parameters, switch to the VelocityJointInterface or EffortJointInterface, or upgrade to Gazebo 9.
[ WARN] [1542163123.906064893, 0.160000000]: For details, see https://github.com/ros-simulation/gazebo_ros_pkgs/issues/612
[spawn_gazebo_model-4] process has finished cleanly
log file: /home/zhekai/.ros/log/65e4e344-e7b6-11e8-8bb2-cc2f71901a9b/spawn_gazebo_model-4*.log
[WARN] [1542163172.767687, 48.739000]: Controller Spawner couldn't find the expected controller_manager ROS interface.

It seems that I have loaded gazebo_ros_control, but the controller goes wrong.

geweihgg, 2018-11-14:
Hi, @zhekaiTong ,
It seems that I've encountered this problem. If I'm not mistaken, you can check the controller type. 
In UR package, it provides an example of combining UR5 and robotiq gripper, but the controller type is JointTrajectoryController. And we can find that JointPositionController is employed in robotiq_85.launch. So, maybe you can load gripper_trajectory_controller_robotiq.yaml in robotiq_85.launch to replace JointPostionController with JointTrajectoryController.
GeWei

geweihgg, 2018-11-14:
Hi, @zhekaiTong ,
Dose it work?

zhekaiTong, 2018-11-14:
Hi, @geweihgg ,

It still does not work. Could you please tell me where I can find the example combining UR5 and robotiq gripper, and gripper_trajectory_controller_robotiq.yaml file? Currently, my controller.yaml file is like this:
`ur10_gripper_controller:
  type: position_controllers/JointTrajectoryController
  joints:
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint
     - gripper_finger1_joint
  constraints:
      goal_time: 0.6
      stopped_velocity_tolerance: 0.05
      shoulder_pan_joint: {trajectory: 0.1, goal: 0.1}
      shoulder_lift_joint: {trajectory: 0.1, goal: 0.1}
      elbow_joint: {trajectory: 0.1, goal: 0.1}
      wrist_1_joint: {trajectory: 0.1, goal: 0.1}
      wrist_2_joint: {trajectory: 0.1, goal: 0.1}
      wrist_3_joint: {trajectory: 0.1, goal: 0.1}
      gripper_finger1_joint: {trajectory: 0.1, goal: 0.1}
  stop_trajectory_duration: 0.5
  state_publish_rate:  25
  action_monitor_rate: 10`

It seems that my controller type is already JointTrajectroyController. Do you have any idea? Thanks a lot!

zhekaiTong, 2018-11-14:
@geweihgg Sorry for the format, the code may look better in this way:
> ur10_gripper_controller:
  type: position_controllers/JointTrajectoryController
  joints:
     - shoulder_pan_joint
     - shoulder_lift_joint
     - elbow_joint
     - wrist_1_joint
     - wrist_2_joint
     - wrist_3_joint
     - gripper_finger1_joint
  constraints:
      goal_time: 0.6
      stopped_velocity_tolerance: 0.05
      shoulder_pan_joint: {trajectory: 0.1, goal: 0.1}
      shoulder_lift_joint: {trajectory: 0.1, goal: 0.1}
      elbow_joint: {trajectory: 0.1, goal: 0.1}
      wrist_1_joint: {trajectory: 0.1, goal: 0.1}
      wrist_2_joint: {trajectory: 0.1, goal: 0.1}
      wrist_3_joint: {trajectory: 0.1, goal: 0.1}
      gripper_finger1_joint: {trajectory: 0.1, goal: 0.1}
  stop_trajectory_duration: 0.5
  state_publish_rate:  25
  action_monitor_rate: 10

geweihgg, 2018-11-14:
@zhekaiTong ,
Sorry for my mistake. **gripper_trajectory_controller_robotiq.yaml** is created by myself. You can create this file, and fill it with these codes:
`gripper:
  type: position_controllers/JointTrajectoryController
  joints:
     - robotiq_85_left_knuckle_joint
  constraints:
      goal_time: 0.6
      stopped_velocity_tolerance: 0.05
      robotiq_85_left_knuckle_joint: {trajectory: 0.1, goal: 0.1}
  stop_trajectory_duration: 0.5
  state_publish_rate:  25
  action_monitor_rate: 10`

Have a try.

zhekaiTong, 2018-11-14:
@geweihgg @LuserBen Hi, I solve the warning

> [WARN] [1542184004.169744, 29.097000]: Controller Spawner couldn't find the expected controller_manager ROS interface.
It's a namespace incompatibility issue.

1. In .urdf/.xacro file, add gazebo tag with gazebo_ros_control plugin: 
`<gazebo>`
        `<plugin name="gazebo_ros_control" filename="/opt/ros/kinetic/lib/libgazebo_ros_control.so">`
            `<robotNamespace>/robotiq_85_gripper</robotNamespace>`
        `</plugin>`
`</gazebo>`

2. In robotiq_85.launch where you launch spawner and controller, add namespace "ns=" to both controller.yaml tag and controller manager spawner tag, like this:
`<rosparam file="$(find robotiq_85_gazebo)/controller/gripper_controller_robotiq.yaml" ns="/robotiq_85_gripper" command="load"/>`
`<node name="gripper_controller_spawner" pkg="controller_manager" type="spawner" args="gripper" ns="/robotiq_85_gripper" respawn="false" output="screen" />`

Thanks.

LuserBen, 2018-11-14:
@zhekaiTong 
Oh, nice, 666.
I did not have this waring so it is no problem for me. I use ur5 and robotiq in indigo.

geweihgg, 2018-11-14:
@zhekaiTong 
Congratulations!