StanleyInnovation/robotiq_85_gripper #17: Joint not connected in Rviz

philwall3, 2018-01-11:
Hi,
as seen in the screenshot, Rviz has trouble with one joint. It's not connected there. Any ideas what could cause this?
[image]

LuserBen, 2018-03-29:
I met the same issue.
Maybe You should only choose one joint (not both finger1_joint and finger2_joint) when set end effector group in moveit_setup_assistant.

PatrickHussey, 2018-03-29:
Finger 2 uses the mimic joint plugin and should be set to passive in the moveit config. But both should be added to the gripper group. Controllers should be for finger 1 only

LuserBen, 2018-03-29:
Ok . Thanks. This is just what I ignore. @PatrickHussey