StanleyInnovation/robotiq_85_gripper #21: Robotiq_85_gripper cann't work in gazebo

scliu2018, 2018-06-08:
Robotiq_85_gripper in gazebo always shock and the joints broke. But the robot can work well in rviz. I don't why.
[image]

ShinYuFish, 2018-12-10:
Hi, you can check your `robotiq_85_gripper.transmission.xacro` the `multiplier` of joint 4 and 5 are set to -1.0 or not.

Mo-637, 2019-02-16:
Have you found a solution for this issue ?
We've facing this same issue,
The multipliers for joints 4 and 5 are set tp -1 we've also tried to set pid gains for each joint to overcome gravity and stop the breaking apart, we've also tried adding friction to the joints which only makes the fall slower, but nothing seems to work

ShinYuFish, 2019-02-18:
Have you checked the <multiplier> parameter of both `robotiq_85_gripper.urdf.xacro` and `robotiq_85_gripper.xacro` files in `robotiq_85_description`?

Also, does your gripper broken without grasping anything or it broken when grasping?

Mo-637, 2019-02-18:
The parameters in the urdfs and xacro seem correct.
The problem is that no finger-finger_tip joint is defined, and it can't be defined since urdfs don't handle closed chains gracefully.
Our problem seemed to be fixed by changing the gripper controller to a GripperActionController, and using the mimic plugin.

Mo-637, 2019-02-18:
It used to break without grasping, or any active controllers sending commands, the links fell under their own weights

PratNag, 2019-02-22:
Hello @Mo-637 I am facing the exact same problem. Does it work if you change to GripperActionController? Can you explain/share files how I do that?

Mo-637, 2019-02-22:
In the gripper controller .yaml file
Change the controller type to a GripperActionController instead of a Jointtrajectorycontroller

PratNag, 2019-02-22:
Did that as well changed the gripper controller yaml file as well. It still behaves as it earlier.

zhangkefang45, 2019-03-19:
@Mo-637 Hi, when I change the controller type to GripperActionController instead of a Jointtrajectorycontroller, I got the error: 
[ERROR] [1552964566.780981362, 0.471000000]: Could not find joint name on param server
[ERROR] [1552964566.781095273, 0.471000000]: Failed to initialize the controller
[ERROR] [1552964566.781148702, 0.471000000]: Initializing controller 'gripper' failed
How to fix that?

zhangkefang45, 2019-03-19:
Looks like my yaml file wrong.  I change my gripper_controller.yaml to:
gripper:
  type: position_controllers/GripperActionController
  joint: simple_gripper_robotiq_85_left_knuckle_joint
  action_monitor_rate: 20
  goal_tolerance: 0.002
  max_effort: 100
  stall_velocity_threshold: 0.001
  stall_timeout: 1.0
the gripper can close, but the joint still spin...I dont't know why.

L-eonor, 2021-03-28:
I had the same issue. The problem was at the gazebo mimic joint plugin configuration. The multipliers params should be either 1 or -1 as a reference to the joint being mimicked.

HBRoad, 2021-04-09:
> I had the same issue. The problem was at the gazebo mimic joint plugin configuration. The multipliers params should be either 1 or -1 as a reference to the joint being mimicked.

Hi, I have the same issues, and I have set the multipliers params to -1 in every mimic joint, but nothing changed, can you tell me which file should be changed? I want to make sure mine is right. Thank you so much!

L-eonor, 2021-04-09:
Hello @HBRoad , I've a currently open PR in robotiq repo that solves the issue: https://github.com/ros-industrial/robotiq/pull/186

HBRoad, 2021-04-10:
> Hello @HBRoad , I've a currently open PR in robotiq repo that solves the issue: [ros-industrial/robotiq#186](https://github.com/ros-industrial/robotiq/pull/186)

Hi, it's still not working, my ubuntu version is 18.04(melodic), and gazebo version is 9.16.0, is this working on melodic?

HBRoad, 2021-04-10:
> Hello @HBRoad , I've a currently open PR in robotiq repo that solves the issue: [ros-industrial/robotiq#186](https://github.com/ros-industrial/robotiq/pull/186)

I have tested on ubuntu 16.04 and gazebo 7.16, everything works fine, I don't know why it's not working on 18.04

L-eonor, 2021-04-10:
> > Hello @HBRoad , I've a currently open PR in robotiq repo that solves the issue: [ros-industrial/robotiq#186](https://github.com/ros-industrial/robotiq/pull/186)
> 
> Hi, it's still not working, my ubuntu version is 18.04(melodic), and gazebo version is 9.16.0, is this working on melodic?

yes

HBRoad, 2021-04-10:
> > > Hello @HBRoad , I've a currently open PR in robotiq repo that solves the issue: [ros-industrial/robotiq#186](https://github.com/ros-industrial/robotiq/pull/186)
> > 
> > 
> > Hi, it's still not working, my ubuntu version is 18.04(melodic), and gazebo version is 9.16.0, is this working on melodic?
> 
> yes

Well, maybe something wrong with mine, which I don't know yet, thank you anyway.

L-eonor, 2021-04-10:
did you install the mimic joint plugin?

> > Hello @HBRoad , I've a currently open PR in robotiq repo that solves the issue: [ros-industrial/robotiq#186](https://github.com/ros-industrial/robotiq/pull/186)
> 
> I have tested on ubuntu 16.04 and gazebo 7.16, everything works fine, I don't know why it's not working on 18.04