StanleyInnovation/robotiq_85_gripper #22: Errors when buidling

IndustrialEngStudent, 2019-05-23:
Hi, I would really like to use this package. However, I get 5 pages full of errors when I try to build it. 

The errors start here after `catkin_make`:

```
up_gazebo_plugins/src/disable_link_plugin.cpp:23:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
In file included from /usr/include/c++/5/random:35:0,
                 from /usr/include/ignition/math2/ignition/math/Rand.hh:20,
                 from /usr/include/ignition/math2/ignition/math.hh:18,
                 from /usr/include/sdformat-4.0/sdf/Param.hh:34,
                 from /usr/include/sdformat-4.0/sdf/Element.hh:24,
                 from /usr/include/sdformat-4.0/sdf/sdf.hh:5,
                 from /usr/include/gazebo-7/gazebo/common/Plugin.hh:42,
                 from /home/laurah/catkin_ws/src/robotiq_85_gripper/robotiq_85_simulation/roboticsgroup_gazebo_plugins/include/roboticsgroup_gazebo_plugins/mimic_joint_plugin.h:36,
                 from /home/laurah/catkin_ws/src/robotiq_85_gripper/robotiq_85_simulation/roboticsgroup_gazebo_plugins/src/mimic_joint_plugin.cpp:23:
/usr/include/c++/5/bits/c++0x_warning.h:32:2: error: #error This file requires compiler and library support for the ISO C++ 2011 standard. This support must be enabled with the -std=c++11 or -std=gnu++11 compiler options.
 #error This file requires compiler and library support \
  ^
[ 62%] Generating Python msg __init__.py for robotiq_85_msgs
[ 63%] Generating C++ code from
```

Then, later on, I get multiple errors ike this:

```
[100%] Built target rq_sensor
In file included from /usr/include/sdformat-4.0/sdf/sdf.hh:3:0,
                 from /usr/include/gazebo-7/gazebo/common/Plugin.hh:42,
                 from /home/laurah/catkin_ws/src/robotiq_85_gripper/robotiq_85_simulation/roboticsgroup_gazebo_plugins/include/roboticsgroup_gazebo_plugins/disable_link_plugin.h:33,
                 from /home/laurah/catkin_ws/src/robotiq_85_gripper/robotiq_85_simulation/roboticsgroup_gazebo_plugins/src/disable_link_plugin.cpp:23:
/usr/include/sdformat-4.0/sdf/Console.hh:55:16: error: ‘shared_ptr’ in namespace ‘std’ does not name a template type
   typedef std::shared_ptr<Console> ConsolePtr;
                ^
/usr/include/sdformat-4.0/sdf/Console.hh:96:20: error: ‘ConsolePtr’ does not name a type
     public: static ConsolePtr Instance();
                    ^
/usr/include/sdformat-4.0/sdf/Console.hh:118:19: error: ‘unique_ptr’ in namespace ‘std’ does not name a template type
     private: std::unique_ptr<ConsolePrivate> dataPtr;
                   ^
/usr/include/sdformat-4.0/sdf/Console.hh: In member function ‘sdf::Console::ConsoleStream& sdf::Console::ConsoleStream::operator<<(const T&)’:
/usr/include/sdformat-4.0/sdf/Console.hh:145:9: error: ‘Instance’ is not a member of ‘sdf::Console’
     if (Console::Instance()->dataPtr->logFileStream.is_open())
         ^
/usr/include/sdformat-4.0/sdf/Console.hh:147:7: error: ‘Instance’ is not a member of ‘sdf::Console’
       Console::Instance()->dataPtr->logFileStream << _rhs;
       ^
/usr/include/sdformat-4.0/sdf/Console.hh:148:7: error: ‘Instance’ is not a member of ‘sdf::Console’
       Console::Instance()->dataPtr->logFileStream.flush();
       ^
In file included from /usr/include/ignition/math2/ignition/math/AffineException.hh:21:0,
                 from /usr/include/ignition/math2/ignition/math.hh:3,
                 from /usr/include/sdformat-4.0/sdf/Param.hh:34,
                 from /usr/include/sdformat-4.0/sdf/Element.hh:24,
                 from /usr/include/sdformat-4.0/sdf/sdf.hh:5,
                 from /usr/include/gazebo-7/gazebo/common/Plugin.hh:42,
                 from /home/laurah/catkin_ws/src/robotiq_85_gripper/robotiq_85_simulation/roboticsgroup_gazebo_plugins/include/roboticsgroup_gazebo_plugins/disable_link_plugin.h:33,
                 from /home/laurah/catkin_ws/src/robotiq_85_gripper/robotiq_85_simulation/roboticsgroup_gazebo_plugins/src/disable_link_plugin.cpp:23:
/usr/include/ignition/math2/ignition/math/Helpers.hh: In function ‘T ignition::math::precision(const T&, const unsigned int&)’:
/usr/include/ignition/math2/ignition/math/Helpers.hh:242:14: error: ‘round’ is not a member of ‘std’
       return std::round(_a * pow(10, _precision)) / pow(10, _precision);

```
I do not really understand what is happening, what am I missing?

PatrickHussey, 2019-05-23:
should be all set now

closed 2019-05-23 by PatrickHussey

IndustrialEngStudent, 2019-05-24:
Thank you, no errors this time! :)