StanleyInnovation/robotiq_85_gripper #24: stop gripper from closing in robotiq_85.launch

IndustrialEngStudent, 2019-06-13:
Hi,

I want to bring up the robotiq_85_gripper with the robotiq_85_driver, without the gripper starting to close and open. I want the gripper to hold something (61 mm width) before and after `roslaunch robotiq_85_driver robotiq_85.launch `. Right know everytime I launch the gripper closes and openes to 85 mm. 

I tried to look into the robotiq_85_driver/src/*.py files. However, I don't know how to alter it such that the gripper stays still when launching `robotiq_85.launch `. 
I would be very grateful if someone could help me :)

PatrickHussey, 2019-06-13:
Unfortunately that's a behavior of the gripper startup. When you activate the gripper it goes to the endpoints (probably to calibrate its position). And you have to activate it before you can command it as far as I know.

IndustrialEngStudent, 2019-06-13:
Ah okay. In that case, I will attach the object to the gripper after `robotiq_85.launch` every time I start the gripper. Thank you for the fast response!:)

closed 2019-08-01 by IndustrialEngStudent